nliaudat/esp_canbus #2: help needed?

bagges, 2023-06-01:
Did you already try the [hoval_HV.yaml](https://github.com/nliaudat/esp_canbus/blob/main/esphome/hoval_HV.yaml). Do you get data? Is there something I can help?

nliaudat, 2023-06-01:
It's not working actually.
Cannot compile since the can_id is not the datapointID from hoval documentation.

bagges, 2023-06-01:
Okay. I am also not sure what https://github.com/nliaudat/esp_canbus/blob/d2fb3f1be5307dc7e3fb5344d68a3526cb0d8b73/esphome/hoval_HV.yaml#L152 should be. Is it the same like: https://github.com/chrishrb/hoval-gateway/blob/ac464a2df67cab42e2ad534dceed7287fbdec3ae/settings.yml#L9

chrishrb, 2023-06-03:
Hi guys,
is the can_id (esphome can) the same as the arbitration_id (python can package)? If yes (and i believe this is the case here), then the can_id (on the screenshot it's the arbitration_id) is created by the following data (as far as i know, its only reverse engineered):

[image]

I hope this helps you two! At the moment I don‘t have the time to try this board at home and help you with the implementation, but if you have more specific questions regarding this drawing and how the hoval can bus works, don‘t hesitate to ask! 👍