nliaudat/esp_canbus #6: unable to compile

bagges, 2023-10-20:
I was not able to compile the config.

```
esphome compile config.yaml
INFO Reading configuration config.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

captive_portal: [source wifi.yaml:18]

  This feature is only available with frameworks ['arduino'].
  {}
web_server: [source config.yaml:41]

  This feature is only available with frameworks ['arduino'].
  port: 80
async_tcp: None
  {}
This feature is only available with frameworks ['arduino']
```

I changed the `esp32` section to:

```
esp32:
  board: nodemcu-32s
  framework:
    type: arduino #arduino #esp-idf  not ready missing json and async_tcp
    version: recommended #latest
```

Now I was able to compile. Did not test yet if it's working.

nliaudat, 2023-10-21:
webserver and async_tcp are available only with latest esphome and esp-idf framework. 

pip install --upgrade esphome
pip install --upgrade platformio

bagges, 2023-10-21:
> webserver and async_tcp are available only with latest esphome and esp-idf framework.
> 
> pip install --upgrade esphome pip install --upgrade platformio

update to latest version fixed the problem. thanks.

closed 2023-10-21 by bagges