nliaudat/esp_canbus #7: Composition of abitration ID

abichinger, 2023-12-15:
Hey!

Thank you for sharing this awesome project.
I'm currently working on adapting it for a heat generator. (https://github.com/abichinger/esp_canbus)

I'm trying to figure out how the can id is composed and thought you might be able to help me with this.
https://github.com/chrishrb/hoval-gateway/issues/11

nliaudat, 2023-12-15:
Hi, 

Feel free to make a pull request if you made improvements !

I do not understand your request cause in your title you says "arbitration" and in the text "can ID". That's not exactly the same. 

With esphome, you haven't to care about arbitration ID, the canbus library do the job for you. 

You only have to give the can_id aka "the channel" to ear and can_id_mask (to filter one or perhaps many devices on the same channel)

```
can_id: ${hoval_homevent_can_Addr} #0x1FD047FF #// Set filter to accept only the frame with identifier 1FD047FF (hoval homevent)
can_id_mask: 0x1FFFFFFF #// 0x1FFFFFFF, which means that all bits of the received CAN ID are compared to the corresponding bits of 0x1FD047FF. Since all bits are set to 1 in can_id_mask, the received frame is accepted only if its CAN ID matches 0x1FD047FF exactly
```

_Messages are labeled by an identifier (can ID) assigned to one or more nodes on the network. All nodes receive the message and perform a filtering operation. That is, each node executes an acceptance test on the identifier to determine if the message, and thus its content, is relevant to that particular node. Only the node(s) for which the message is relevant will process it. All others ignore the message.

The identifier has two more functions, as well. It contains data that specifies the priority of the message and it allows the hardware to arbitrate for the bus. That is, it’s used to determine which node gets to transmit if several nodes attempt to do so simultaneously. Message IDs must be unique in a single CAN Bus network, otherwise two nodes would continue transmission beyond the end of the arbitration field (ID) causing an error. The lower the numerical ID, the higher the message priority._

Here you'll find documentation about arbitration ID : 

- https://copperhilltech.com/blog/controller-area-network-can-bus-bus-arbitration/
- https://www.linkedin.com/pulse/bus-arbitration-can-controller-area-network-abdul-rehman-anwer/
- https://www.mikroe.com/blog/can-bus
- https://www.ti.com/lit/an/sloa101b/sloa101b.pdf?ts=1702652659084&ref_url=https%253A%252F%252Fwww.google.com%252F

abichinger, 2023-12-15:
Thank you for the quick response.

> Feel free to make a pull request if you made improvements!

I've rewrote most of it, but I'll open a PR as soon as I'm done.

> I do not understand your request cause in your title you says "arbitration" and in the text "can ID". That's not exactly the same.

I called it abitration ID, because [chrishrb/hoval-gateway](https://github.com/chrishrb/hoval-gateway) calls it abitration ID (most of the time it is called can ID or identifier). To me it seems to be the same thing, because the can ID is used for abitration. The can ID of esphome has a lenght of [11 or 29 Bit](https://github.com/esphome/esphome/blob/836a3db163b130a4aecf3a632868dc3a90c609c8/esphome/components/canbus/canbus.h#L60C36-L60C44). This fits the documentation about arbitration ID that you linked.

Nevertheless, I'm actually interested in how Hoval puts together the can ID.
For example, I'm wondering whether you can extract the device type from the can ID, so that you can automatically determine which Hoval device it is.

nliaudat, 2023-12-16:
I mean the can ID are hardcoded. 
You can find explainations in http://www.hoval.com/misc/TTE/TTE-GW-Modbus-datapoints.xlsx

abichinger, 2023-12-16:
> I mean the can ID are hardcoded.

I think you can modify the can ID of hoval devices with a DIP-Switch.

> You can find explainations in http://www.hoval.com/misc/TTE/TTE-GW-Modbus-datapoints.xlsx

Thanks! I looked at the datasheet from hoval and finally figured it out.

The module address is placed inside the can id from bit 21 to 11.

e.g.:
The start address of a control module is 1025.
My control has the address 18. I got this value from the GUI of the module.

=> Module address = 1025 + (18 - 1) = 1042 = 10 0000 1001 0 binary

Can ID of my TTE-BM (control module): `0x1FE09001` =  0001 1111 11**10 0000 1001 0**000 0000 0001 binary

closed 2023-12-16 by abichinger

nliaudat, 2023-12-17:
You can generate all the needed addresses for your heat genarator by using https://github.com/nliaudat/esp_canbus/tree/main/esphome/hoval_data_processing

nliaudat, 2023-12-17:
Have you tried https://github.com/nliaudat/esp_canbus/wiki/Find-Hoval-canbus-addresses ?

abichinger, 2023-12-18:
Yes, that is how I found the addresses. :+1: