nliaudat/esp_canbus #9: hardcoded canID in new version

nliaudat, 2024-02-06:
https://github.com/nliaudat/esp_canbus/blob/main/esphome/components/toptronic/toptronic.cpp
line 194

`    uint32_t can_id = 0xF1E40801;`

abichinger, 2024-02-07:
Does this `can_id` not work for you?

nliaudat, 2024-02-07:
> Does this `can_id` not work for you?

It's working partially
Not all informations from HV are reported. 

The inputs are not working.

abichinger, 2024-02-07:
Have you tested the latest version including the commit "Build dynamic can_id"?

abichinger, 2024-02-07:
Thank you for the invitation :)

nliaudat, 2024-02-07:
Every things working great !
(have you tried the user level code ?)

abichinger, 2024-02-07:
> Every things working great !
Nice!
> have you tried the user level code ?
Not yet. What can you do if you unlock the control module?

nliaudat, 2024-02-07:
> > have you tried the user level code ?
> 
> Not yet. What can you do if you unlock the control module?

You can modify all parameters. :). 
But the code is perhaps not working as it can be modified by your resealer. 

You can try to find yours with :

######## get users passwords ###########
```

canbus:
  - platform: esp32_can
    id: cbus_psw
    tx_pin: ${can_tx_pin} # GPIO22 #GPIO5
    rx_pin: ${can_rx_pin} # GPIO21 #GPIO4
    can_id: ${TTE_BM_can_id}
    bit_rate: 50kbps # real rate of hoval canbus is 0.5 kbps. Esphome with Esp32 cannot run under 50kbps
    on_frame:
    - can_id: 0
      can_id_mask: 0
      use_extended_id: true
      then:
      - lambda: |-
            //ESP_LOGI("cbus_psw", "%x : %x %x %x %x %x %x %x %x", can_id, int(x[0]), int(x[1]), int(x[2]), int(x[3]), int(x[4]), int(x[5]), int(x[6]), int(x[7]));
            if(x[1] == 0x42 && x[2] == 0x59) { // response frame
                ESP_LOGI("cbus_psw", "%x : %x %x %x %x %x %x %x %x", can_id, int(x[0]), int(x[1]), int(x[2]), int(x[3]), int(x[4]), int(x[5]), int(x[6]), int(x[7]));
            }

select: 
  - platform: template
    name: "Get users passwords"
    id: canbus_get_psw
    optimistic: true
    initial_option: "None"
    options:
      - "None"
      - "GET-user-password-1"
      - "GET-user-password-2"
      - "GET-user-password-3"
      - "GET-user-password-4"
      - "GET-user-password-5"
      - "GET-user-password-6"
      - "GET-user-password-7"
    on_value:
      then: #data : message_len = 0x01, operation = 0x40, function_group = 89 (0x59), function_number = 1 (0x01), datapoint = 1 (0x01) to 7
        - lambda: |-
            uint8_t user = 0xFF;
            if ( x == "GET-user-password-1") {
              user = 0x01;
            }else if ( x == "GET-user-password-2") {
              user = 0x02;
            }else if ( x == "GET-user-password-3") {
              user = 0x03;
            }else if ( x == "GET-user-password-4") {
              user = 0x04;
            }else if ( x == "GET-user-password-5") {
              user = 0x05;
            }else if ( x == "GET-user-password-6") {
              user = 0x06;
            }else if ( x == "GET-user-password-7") {
              user = 0x07;
            }
            uint32_t can_id = ${hoval_toptronic_can_Addr}; //0x1FE04208;
            bool use_extended_id = 1;
            std::vector< uint8_t > data{0x01, 0x40, 0x59, 0x01, user};
            id(cbus_psw)->send_data(can_id, use_extended_id, data);   
```

abichinger, 2024-02-07:
Thank you! The heating generator belongs to my father. I'll ask him if he needs the user level codes.

nliaudat, 2024-02-07:
The components/toptronic is yours, 
Feel free to publish it on esphome official components for the community.

I can name you as code owner on that repo if you want.
(I don't care about anything except the good of the contributors and the community.)

with "user level code", you can break warranty or working stuff. But many datapoints are not documented in TTE-GW-Modbus-datapoints.xlsx

closed 2024-02-07 by nliaudat

abichinger, 2024-02-07:
> The components/toptronic is yours,
Feel free to publish it on esphome official components for the community.

I've already thought about it, maybe I'll do it at some point

> I can name you as code owner on that repo if you want.

No it is fine. I'm glad I'm in the list of contributors.

nliaudat, 2024-02-07:
Many thanks to YOU, a real contributor with skills.