nliaudat/esp_canbus #13: Unique entity id

bagges, 2024-02-08:
What do you think about entity id's with a unique device identifier in it? With the current setup you get not so meaningful entity ids which makes it hard to find them in home assistant during dashboard or automation creation.

For example:

[image]

With the use of friendly_name or some other variable it would be easier to identify the sensors afterwards.

nliaudat, 2024-02-09:
The id is already unique : 

 **id: HV_50_0_0**
which is Hardware / function group / function number / datapoint

```
- platform: toptronic
  name: Outside air temp.
  device_type: HV
  device_addr: ${TT_HV_addr}
  function_group: 50
  function_number: 0
  datapoint: 0
  id: HV_50_0_0
  type: S16
  unit_of_measurement: "\xB0C"
  accuracy_decimals: 1
  filters:
  - multiply: 0.1
```

What you want is adding a friendly_name ?

```
- platform: toptronic
  name: $friendly_name - Outside air temp.
```

The actual output will be : CanBus Controller - Outside air temp.

abichinger, 2024-02-09:
Here is the documentation about [Friendly Naming](https://www.home-assistant.io/integrations/esphome/#friendly-naming).

Example:

```yaml
esphome:
  name: "canbus"
  friendly_name: "CanBus Controller"

sensor:
  name: "Outside air temp."
```

The HA entity would be named `CanBus Controller Outside air temp.` and have an entity ID of `sensor.canbus_outside_air_temp`

Personally I prefer the entity name without the friendly_name prefix.

In my opinion the ideal entity ID would be `sensor.canbus_hv_18_50_0_0` (18 is `TT_HV_addr`) with the entity name being `Outside air temp.`. This would allow to rename the sensor without losing the recorded data. Unfortunately this combination is not possible (https://github.com/esphome/feature-requests/issues/2119).

bagges, 2024-02-10:
setting friendly_name in the esphome section was what I searched for. thanks.

closed 2024-02-10 by bagges