nliaudat/esp_canbus #14: u32 values

andresp, 2024-06-04:
Trying to figure out how to retrieve U32 based high/low values such as these datapoints for a heat pump:

[image]

I know the data is reported to the displays on the unit, but I don't see them going over the CAN bus. I can only  see the GET request data going over the wire:

```
19:22:57	[I]	[tt:230]	
[GET] Data: 0x01400a0159e1
```

Could it be that the response neither fit the SENSOR nor TEXTSENSOR format? Has anyone successfully read u32 based datapoints in an actual environment?

nliaudat, 2024-06-04:
It's not implemented now as U32 and S32 values are only readable and are a very small part of all Toptronic datapoints. 

In fact the canbus communication of hoval devices is a mix of canbus and modbus.

The U32 and S32 values are splited in 2 messages over canbus if my memory is good. 
They should be regrouped to get the raw value. 

I have no time to support you further for now.  Very sorry.

Perhaps @abichinger could help you

Regards

nliaudat, 2024-06-04:
You can open a discussion. 
I mean is not a real issue but more a feature request.
You can close it if you agree

andresp, 2024-06-05:
Happy to change this to a feature request. I could also help with contributing an implementation for the split messages, if you happen to know how to identify the messages that belong together. I’ve been unable to identify these, despite having a few wireshark dumps of the can bus messages. If you could share what to look for, I could probably put this together.