nliaudat/esp_canbus #16: Add support for active error

Dieghito72, 2024-08-06:
**Is your feature request related to a problem? Please describe.**
I have a HOVAL VMC with only the basic controller, not the TopTronic E controller. So I use an ESP with this component to control and monitor the VMC.
I would like also to monitor error codes from the VMC.

[image]

**Describe the solution you'd like**
If I understand well at the moment this is not possible because error info are spread across multiple bytes (15) so there isn't a data type that can contains all the information, so even creating a single sensor containing all the values is not possible.

The idea would be having different sensors (maybe a new sensor type?) were is possible to define start byte:

```YAML
sensor:
- platform: toptronic
  name: Active error 1 appearance timestamp
  device_type: HV
  device_addr: ${TT_HV_addr}
  function_group: 0
  function_number: 0
  datapoint: 29042
  start_byte: 0
  id: HV_0_0_29042
  type: U32
- platform: toptronic
  name: Active error 1 disappear timestamp
  device_type: HV
  device_addr: ${TT_HV_addr}
  function_group: 0
  function_number: 4
  datapoint: 29042
  start_byte: 4
  id: HV_0_0_29042
  type: U32
```

nliaudat, 2024-08-07:
Hi, 
Since there is no documentation on the _**Type**_ meaning, it will be difficult to handle it. 
I do not think the _**Type**_ means bytes. 
I mean it's the number of the message sequence as for U32 values https://github.com/nliaudat/esp_canbus/discussions/15
But, it's only the 3rd to 11th message ?

Keep in mind that it's difficult to implement new stuff on a reverse engineered protocol (mixed of canbus and modbus).

I cannot produce errors on my device. Do you have a candump for studies ?

---- DataPoints ----

```
UnitName | UnitId | FunctionGroup | FunctionNumber | DatapointId | DatapointName | **_Type_** | TypeName
HV | 520 | 0 | 0 | 29042 | Active error 1_appearance_time | 3 | U16
HV | 520 | 0 | 0 | 29042 | Active error 1_appearance_date | 4 | U16
HV | 520 | 0 | 0 | 29042 | Active error 1_disappear_time | 5 | U16
HV | 520 | 0 | 0 | 29042 | Active error 1_disappear_date | 6 | U16
HV | 520 | 0 | 0 | 29042 | Active error 1_source | 7 | U16
HV | 520 | 0 | 0 | 29042 | Active error 1_function_group | 8 | U8
HV | 520 | 0 | 0 | 29042 | Active error 1_function_number | 9 | U8
HV | 520 | 0 | 0 | 29042 | Active error 1_error_type | 10 | U8
HV | 520 | 0 | 0 | 29042 | Active error 1_error_code | 11 | U16
```

and for U32 :

```
HV | 520 | 0 | 0 | 41613 | Cleaning count value message cleaning (operating weeks high | 2 | U32
HV | 520 | 0 | 0 | 41613 | Cleaning count value message cleaning (operating weeks) low | 1 | U32
```

nliaudat, 2024-08-09:
Hi, 
Plz have a try with the new release that can parse U32 & S32 (message with lenght >1)

```
- platform: toptronic
  name: Active error 1
  device_type: HV
  device_addr: ${TT_HV_addr}
  function_group: 0
  function_number: 0
  datapoint: 29042
  id: HV_0_0_29042
  type: S32
  unit_of_measurement: ''
```
I cannot reproduce errors on my HV device.

dkarv, 2024-08-09:
>  If I understand well at the moment this is not possible because error info are spread across multiple bytes (15) so there isn't a data type that can contains all the information, so even creating a single sensor containing all the values is not possible.

I think it is very hard to achieve this in this project. It requires tracking the state of multiple datapoints.
In case you are using home assistant, you could combine them using a template sensor. Also ESPHome allows combining multiple sensors as far as I know...

closed 2024-08-20 by nliaudat